OmniaGit/odooplm #2: Clean install error

fathomtreelimited, 2017-11-10:
On a clean Ubuntu Server 16 install.

```
Error:
Odoo Server Error

Traceback (most recent call last):
  File "/odoo/odoo-server/odoo/tools/convert.py", line 745, in parse
    self._tags[rec.tag](rec, de, mode=mode)
  File "/odoo/odoo-server/odoo/tools/convert.py", line 655, in _tag_record
    id = self.env(context=rec_context)['ir.model.data']._update(rec_model, self.module, res, rec_id or False, not self.isnoupdate(data_node), noupdate=self.isnoupdate(data_node), mode=self.mode)
  File "/odoo/odoo-server/odoo/addons/base/ir/ir_model.py", line 1468, in _update
    record = record.create(values)
  File "/odoo/enterprise/addons/web_studio/models/studio_mixin.py", line 18, in create
    res = super(StudioMixin, self).create(vals)
  File "/odoo/odoo-server/odoo/addons/base/ir/ir_ui_view.py", line 384, in create
    return super(View, self).create(self._compute_defaults(values))
  File "/odoo/odoo-server/odoo/models.py", line 3283, in create
    self._fields[key].determine_inverse(record)
  File "/odoo/odoo-server/odoo/fields.py", line 1078, in determine_inverse
    getattr(records, self.inverse)()
  File "/odoo/odoo-server/odoo/addons/base/ir/ir_ui_view.py", line 253, in _inverse_arch
    view.write(data)
  File "/odoo/enterprise/addons/web_studio/models/studio_mixin.py", line 27, in write
    res = super(StudioMixin, self).write(vals)
  File "/odoo/odoo-server/odoo/addons/base/ir/ir_ui_view.py", line 400, in write
    return super(View, self).write(self._compute_defaults(vals))
  File "/odoo/odoo-server/odoo/models.py", line 3005, in write
    self._write(old_vals)
  File "/odoo/odoo-server/odoo/models.py", line 3158, in _write
    self._validate_fields(vals)
  File "/odoo/odoo-server/odoo/models.py", line 965, in _validate_fields
    check(self)
  File "/odoo/odoo-server/odoo/addons/base/ir/ir_ui_view.py", line 329, in _check_xml
    raise ValidationError(_('Invalid view definition'))
odoo.exceptions.ValidationError: ('Invalid view definition', None)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/odoo/odoo-server/odoo/http.py", line 646, in _handle_exception
    return super(JsonRequest, self)._handle_exception(exception)
  File "/odoo/odoo-server/odoo/http.py", line 307, in _handle_exception
    raise pycompat.reraise(type(exception), exception, sys.exc_info()[2])
  File "/odoo/odoo-server/odoo/tools/pycompat.py", line 87, in reraise
    raise value
  File "/odoo/odoo-server/odoo/http.py", line 683, in dispatch
    result = self._call_function(**self.params)
  File "/odoo/odoo-server/odoo/http.py", line 339, in _call_function
    return checked_call(self.db, *args, **kwargs)
  File "/odoo/odoo-server/odoo/service/model.py", line 97, in wrapper
    return f(dbname, *args, **kwargs)
  File "/odoo/odoo-server/odoo/http.py", line 332, in checked_call
    result = self.endpoint(*a, **kw)
  File "/odoo/odoo-server/odoo/http.py", line 927, in __call__
    return self.method(*args, **kw)
  File "/odoo/odoo-server/odoo/http.py", line 512, in response_wrap
    response = f(*args, **kw)
  File "/odoo/odoo-server/addons/web/controllers/main.py", line 928, in call_button
    action = self._call_kw(model, method, args, {})
  File "/odoo/odoo-server/addons/web/controllers/main.py", line 916, in _call_kw
    return call_kw(request.env[model], method, args, kwargs)
  File "/odoo/odoo-server/odoo/api.py", line 689, in call_kw
    return call_kw_multi(method, model, args, kwargs)
  File "/odoo/odoo-server/odoo/api.py", line 680, in call_kw_multi
    result = method(recs, *args, **kwargs)
  File "<decorator-gen-39>", line 2, in button_immediate_install
  File "/odoo/odoo-server/odoo/addons/base/module/module.py", line 71, in check_and_log
    return method(self, *args, **kwargs)
  File "/odoo/odoo-server/odoo/addons/base/module/module.py", line 438, in button_immediate_install
[... 49 lines cut ...]
  File "/odoo/odoo-server/odoo/addons/base/ir/ir_ui_view.py", line 329, in _check_xml
    raise ValidationError(_('Invalid view definition'))
odoo.tools.convert.ParseError: "Invalid view definition
None" while parsing /odoo/odooplm/plm/views/mrp_extension.xml:190, near
<record id="view_mrp_bom_filter_inherit" model="ir.ui.view">
            <field name="name">plm.bom.select</field>
            <field name="model">mrp.bom</field>
         	<field name="inherit_id" ref="mrp.view_mrp_bom_filter"/>
            <field name="type">search</field>
            <field name="arch" type="xml">
                <data><field name="code" position="before">
	               	<filter name="filter_latest_2_weeks" string="Latest 2 Weeks" separator="1" domain="[('create_date','&lt;=', (datetime.date.today()).strftime('%%Y-%%m-%%d')),('create_date','&gt;=',(datetime.date.today()-datetime.timedelta(weeks=2)).strftime('%%Y-%%m-%%d'))]" help="Created latest 2 Weeks"/>
	               	<filter name="filter_latest_7" string="Latest 7 Days" separator="1" domain="[('create_date','&lt;=', (datetime.date.today()).strftime('%%Y-%%m-%%d')),('create_date','&gt;=',(datetime.date.today()-datetime.timedelta(weeks=1)).strftime('%%Y-%%m-%%d'))]" help="Created latest 7 Days"/>
	             	<filter name="filter_yesterday" string="Yesterday" separator="1" domain="[('create_date','&lt;=', (datetime.date.today()-datetime.timedelta(days=1)).strftime('%%Y-%%m-%%d')),('create_date','&gt;=',(datetime.date.today()-datetime.timedelta(days=1)).strftime('%%Y-%%m-%%d'))]" help="Created yesterday"/>
	            	<filter name="filter_today" string="Today" separator="1" domain="[('create_date','&lt;=', (datetime.date.today()).strftime('%%Y-%%m-%%d')),('create_date','&gt;=',(datetime.date.today()).strftime('%%Y-%%m-%%d'))]" help="Created today"/>
	               	<separator orientation="vertical"/>
        		</field>
				<xpath expr="//group" position="after">
                    <filter string="BoM Name" domain="[]" context="{'group_by':'code'}"/>
                    <filter string="Variant Name" domain="[]" context="{'group_by':'product_id'}"/>
                    <filter string="Source Relationship" domain="[]" context="{'group_by':'source_id'}"/>
	            </xpath>
	        </data></field>
        </record>
```

dsmerghetto, 2017-11-13:
Dear Mr. fathomtreelimited,

I tried two times to create a new database installing plm module but I didn't replicate the problem.
Could you check to have latest odoo and plm versions?

Best Regards,
Daniel

closed 2017-11-15 by OmniaGit

SalahAdDin, 2018-06-01:
@dsmerghetto I have the same problem right now:
``` 
odoo_11.web | 2018-06-01 01:19:37,008 1 ERROR rotafilo werkzeug: Error on request:
odoo_11.web | Traceback (most recent call last):
odoo_11.web |   File "/usr/lib/python3/dist-packages/werkzeug/serving.py", line 205, in run_wsgi
odoo_11.web |     execute(self.server.app)
odoo_11.web |   File "/usr/lib/python3/dist-packages/werkzeug/serving.py", line 193, in execute
odoo_11.web |     application_iter = app(environ, start_response)
odoo_11.web |   File "/usr/lib/python3/dist-packages/odoo/service/server.py", line 252, in app
odoo_11.web |     return self.app(e, s)
odoo_11.web |   File "/usr/lib/python3/dist-packages/odoo/service/wsgi_server.py", line 166, in application
odoo_11.web |     return application_unproxied(environ, start_response)
odoo_11.web |   File "/usr/lib/python3/dist-packages/odoo/service/wsgi_server.py", line 154, in application_unproxied
odoo_11.web |     result = handler(environ, start_response)
odoo_11.web |   File "/usr/lib/python3/dist-packages/odoo/http.py", line 1298, in __call__
odoo_11.web |     return self.dispatch(environ, start_response)
odoo_11.web |   File "/usr/lib/python3/dist-packages/odoo/http.py", line 1272, in __call__
odoo_11.web |     return self.app(environ, start_wrapped)
odoo_11.web |   File "/usr/lib/python3/dist-packages/werkzeug/wsgi.py", line 599, in __call__
odoo_11.web |     return self.app(environ, start_response)
odoo_11.web |   File "/usr/lib/python3/dist-packages/odoo/http.py", line 1462, in dispatch
odoo_11.web |     result = ir_http._dispatch()
odoo_11.web |   File "/usr/lib/python3/dist-packages/odoo/addons/auth_signup/models/ir_http.py", line 19, in _dispatch
odoo_11.web |     return super(Http, cls)._dispatch()
odoo_11.web |   File "/usr/lib/python3/dist-packages/odoo/addons/web_editor/models/ir_http.py", line 21, in _dispatch
odoo_11.web |     return super(IrHttp, cls)._dispatch()
odoo_11.web |   File "/usr/lib/python3/dist-packages/odoo/addons/http_routing/models/ir_http.py", line 391, in _dispatch
odoo_11.web |     result = super(IrHttp, cls)._dispatch()
odoo_11.web |   File "/usr/lib/python3/dist-packages/odoo/addons/utm/models/ir_http.py", line 20, in _dispatch
odoo_11.web |     response = super(IrHttp, cls)._dispatch()
odoo_11.web |   File "/usr/lib/python3/dist-packages/odoo/addons/base/ir/ir_http.py", line 212, in _dispatch
odoo_11.web |     return cls._handle_exception(e)
odoo_11.web |   File "/usr/lib/python3/dist-packages/odoo/addons/base/ir/ir_http.py", line 182, in _handle_exception
odoo_11.web |     return request._handle_exception(exception)
odoo_11.web |   File "/usr/lib/python3/dist-packages/odoo/http.py", line 761, in _handle_exception
odoo_11.web |     return super(HttpRequest, self)._handle_exception(exception)
odoo_11.web |   File "/usr/lib/python3/dist-packages/odoo/http.py", line 307, in _handle_exception
odoo_11.web |     raise pycompat.reraise(type(exception), exception, sys.exc_info()[2])
odoo_11.web |   File "/usr/lib/python3/dist-packages/odoo/tools/pycompat.py", line 87, in reraise
odoo_11.web |     raise value
odoo_11.web |   File "/usr/lib/python3/dist-packages/odoo/addons/base/ir/ir_http.py", line 208, in _dispatch
odoo_11.web |     result = request.dispatch()
odoo_11.web |   File "/usr/lib/python3/dist-packages/odoo/http.py", line 820, in dispatch
odoo_11.web |     r = self._call_function(**self.params)
odoo_11.web |   File "/usr/lib/python3/dist-packages/odoo/http.py", line 339, in _call_function
odoo_11.web |     return checked_call(self.db, *args, **kwargs)
odoo_11.web |   File "/usr/lib/python3/dist-packages/odoo/service/model.py", line 97, in wrapper
odoo_11.web |     return f(dbname, *args, **kwargs)
odoo_11.web |   File "/usr/lib/python3/dist-packages/odoo/http.py", line 335, in checked_call
odoo_11.web |     result.flatten()
odoo_11.web |   File "/usr/lib/python3/dist-packages/odoo/http.py", line 1249, in flatten
odoo_11.web |     self.response.append(self.render())
odoo_11.web |   File "/usr/lib/python3/dist-packages/odoo/http.py", line 1242, in render
odoo_11.web |     return env["ir.ui.view"].render_template(self.template, self.qcontext)
odoo_11.web |   File "/usr/lib/python3/dist-packages/odoo/addons/base/ir/ir_ui_view.py", line 1158, in render_template
odoo_11.web |     return self.browse(self.get_view_id(template)).render(values, engine)
odoo_11.web |   File "/usr/lib/python3/dist-packages/odoo/addons/web_editor/models/ir_ui_view.py", line 27, in render
odoo_11.web |     return super(IrUiView, self).render(values=values, engine=engine)
odoo_11.web |   File "/usr/lib/python3/dist-packages/odoo/addons/base/ir/ir_ui_view.py", line 1167, in render
odoo_11.web |     return self.env[engine].render(self.id, qcontext)
odoo_11.web |   File "/usr/lib/python3/dist-packages/odoo/addons/base/ir/ir_qweb/ir_qweb.py", line 57, in render
odoo_11.web |     return super(IrQWeb, self).render(id_or_xml_id, values=values, **context)
odoo_11.web |   File "/usr/lib/python3/dist-packages/odoo/addons/base/ir/ir_qweb/qweb.py", line 271, in render
odoo_11.web |     self.compile(template, options)(self, body.append, values or {})
odoo_11.web |   File "/usr/lib/python3/dist-packages/odoo/addons/base/ir/ir_qweb/qweb.py", line 343, in _compiled_fn
odoo_11.web |     raise e
odoo_11.web |   File "/usr/lib/python3/dist-packages/odoo/addons/base/ir/ir_qweb/qweb.py", line 341, in _compiled_fn
odoo_11.web |     return compiled(self, append, new, options, log)
[... cut ...]
odoo_11.web |                         </a>
odoo_11.web |                     </t>
odoo_11.web |                     <t t-if="not menu['children']">
odoo_11.web |                         <t t-call="web.menu_link"/>
odoo_11.web |                     </t>
odoo_11.web |                     <t t-call="web.menu_secondary_submenu"/>
odoo_11.web |                 </li>
odoo_11.web |             </t>
odoo_11.web | 2018-06-01 01:20:07,381 1 INFO rotafilo werkzeug: 172.18.0.1 - - [01/Jun/2018 01:20:07] "POST /longpolling/poll HTTP/1.1" 200 -
```