OmniaGit/odooplm #3: [11.0]Multiple warnings

SalahAdDin, 2018-06-18:
Hello, i'm testing this module and i'm getting the next errors and warnings:
```bash
2018-06-18 07:14:57,667 15 WARNING rotafilo odoo.modules.registry: Models have no table: report.plm_document.user, report.plm_document.wall, report.plm_files.partner.
2018-06-18 07:14:57,673 15 ERROR rotafilo odoo.modules.registry: Model report.plm_files.partner has no table.
2018-06-18 07:14:57,673 15 ERROR rotafilo odoo.modules.registry: Model report.plm_document.wall has no table.
2018-06-18 07:14:57,673 15 ERROR rotafilo odoo.modules.registry: Model report.plm_document.user has no table.
2018-06-18 07:14:58,166 15 WARNING rotafilo odoo.modules.loading: The model plm.temporary_date_compute has no access rules, consider adding one. E.g. access_plm_temporary_date_compute,access_plm_temporary_date_compute,model_plm_temporary_date_compute,,1,0,0,0
2018-06-18 07:14:58,166 15 WARNING rotafilo odoo.modules.loading: The model document.rev_wizard has no access rules, consider adding one. E.g. access_document_rev_wizard,access_document_rev_wizard,model_document_rev_wizard,,1,0,0,0
2018-06-18 07:14:58,168 15 WARNING rotafilo odoo.modules.loading: The transient model plm.temporary (Temporary Class) should not have explicit access rules!
```

Thank you!

OmniaGit, 2018-09-19:
this need to be fixed

OmniaGit, 2019-04-11:
fixed in v12

closed 2019-04-11 by OmniaGit