OmniaGit/odooplm #6: [Question]Stockable product have no engineering information.

SalahAdDin, 2018-09-11:
We are using this module in an Industry Manufacturing company and i am testing it trying to adapt it to our workflow.

Company manufacture engineering peace that also can be stockable for future sale orders, but i see that only services and consumable products can be related to an Engineering Part.

Why?

I think it is related to the **MO** concept, right?

OmniaGit, 2018-09-19:
as designed only stockable and consumable product could be selected.. engineering shuld accept only this kind of products

regards
Matteo

SalahAdDin, 2018-09-19:
No, stockable products do not handle engineering info. Services and consumables can handle it.

SalahAdDin, 2018-09-21:
I didn't see the button before, i'm sorry.
But, for now, it stills show the button for services.

OmniaGit, 2019-04-11:
fix in v12

closed 2019-04-11 by OmniaGit