OmniaGit/odooplm #9: [11.0][plm_engineering]Engineering BOM does not show CAD item position

SalahAdDin, 2018-09-12:
I'm testing this module for my client, it is very good, but i'm looking for some bugs.

In this case, i have one that i want to call crossing concepts: 
[image]

As you can see in the above animation when you see a normal `BOM`, that does not need a `CAD` item position as well other fields, you will find them in the `BOM` view; while, the Engineering `BOM`, that use `CAD` item positions, does not show that field, why?

That's why i call it crossing concepts: Normal `BOM` shows a logical engineering `BOM`, while Engineering `BOM` shows a normal `BOM`.

Is there any explanation for that?

SalahAdDin, 2018-09-14:
I'm trying to fix this problem; i tried with two or three approaches, but finally i have just to remove and add fields; it has a problem: `consumed in operation` is not added in this view.

And also, it has an interesting issue:
[image]
As you can see, if i get into the view through `Search on Engineering Bill of Materials` i get the custom view, but if i see the `BoM` through the `Engineering Part` view i get the normal component view.

So, why do you use this kind of view? Is better use the common `BoM` view?

SalahAdDin, 2018-09-14:
I think the correct solution is to inherit the `Engine BoM` view from normal `BoM`, but only shows its fields in the proper `Engine BoM` view.

SalahAdDin, 2018-09-19:
Also, i can see you can create normal `BoM` since engineering `BoM` section, it is ...

OmniaGit, 2019-04-11:
fix in v12

closed 2019-04-11 by OmniaGit