OmniaGit/odooplm #13: [plm_suspended] missing un suspend action

SalahAdDin, 2018-09-17:
I am working to update these modules in a proper way to `Odoo 11`, and maybe also `12`; and i found a missing function in the code:
https://github.com/OmniaGit/odooplm/blob/11.0/plm_suspended/views/plm_product_view.xml#L20
it is the same thing in https://github.com/OmniaGit/odooplm/blob/11.0/plm_suspended/views/plm_document_view.xml#L19.

But that functions do not exist in the `python` code, why?

OmniaGit, 2018-09-19:
this module is not finished yet. we just push the modification

SalahAdDin, 2018-09-20:
@OmniaGit oh, really? But i added some improvements, can you review them?

SalahAdDin, 2018-09-21:
@OmniaGit  i'm getting a bug related to this module:
[image]

We can see the register of this actions in the messages:
[image]

In some way, the functions implemented in this module overwrite the behavior of the other button, preventing the states from being changed or reflected in the user interface.
Any help?

Thank you.

OmniaGit, 2019-04-11:
fixed in v12

closed 2019-04-11 by OmniaGit