OmniaGit/odooplm #23: [11.0] it is not possible to manufacture engineering componets

SalahAdDin, 2018-09-24:
I'm trying to manufacture engineering parts but i found a big problem:
[image]
As you can see, when we go to **Manufacturing** section and we create a new **MO**, we cannot see or find engineering parts in the **product**'s field list.

But, on the other side, when we send a **MO** directly from  a **SO** through the _Make to Order_ route, we have no problem, the **SO** creates a **MO** and **PO** for this engineering parts without any problem.

Why happens this?

Thanks

SalahAdDin, 2018-09-24:
Ok, i got it: that field use the next domain: [`domain="[('bom_ids', '!=', False), ('bom_ids.active', '=', True), ('bom_ids.type', '=', 'normal')]"`](https://github.com/odoo/odoo/blob/11.0/addons/mrp/views/mrp_production_views.xml#L82).
It does not take into account other _BoM_ types.
Should we include the other types?

OmniaGit, 2018-09-27:
On the product eng form there is an action that convert the engineering bom in to normal one.
Doing this action you will perform a sum action on the line that are with the some product that came from the cad integration .

closed 2019-04-11 by OmniaGit