OmniaGit/odooplm #25: odooplm 2d print pdf

PaulWang52, 2019-04-06:
Hello

I can't print 2d pdf, how to solve it, thank you 

[image]
[image]
[image]
[image]
[image]

OmniaGit, 2019-04-06:
Dear Paul,
could you please give us more details on how to reproduce the problem ?
The screen shut are not clear
which cad are you using to perform the save operation ?
did you save the file with our integration ?
which version of odoo are you using ?
you get some error ?

regards
Matteo

PaulWang52, 2019-04-06:
Dear Matteo,
Sorry,My English is very poor
I use solid edge
Yes，I use your integration save
Odoo v11
In documents print pdf 2d，but it is blank.

[image]
[image]
[image]
[image]
[image]

closed 2019-04-08 by PaulWang52

OmniaGit, 2019-04-08:
did you solve the problem ?