OmniaGit/odooplm #26: Connector for FreeCAD

Axel-G, 2019-07-02:
I saw a youtube video to connect freecad with your? odooplm.

where can i found this connector?

thx for helping.

OmniaGit, 2019-08-06:
Dear Axel,
the FreedCad integration is under development , there is no packages right now, and we have some issues on it. The integretion will not work out of the box.

if you are still interested i will send you the package

Regards,
Matteo

Axel-G, 2019-08-12:
Hello Matteo,
thx i m looking forward to test your package.
github@r5d.de please

MadeforMaking, 2020-11-13:
Has integration been made yet? I think this would be great with the increase in FreeCAD users!

OmniaGit, 2020-11-13:
Not yet finished .. We are finishing the integration. hopefully we wold like to make it available for the beginning of the 2021 !!

MadeforMaking, 2020-11-14:
Will be looking forward to this. Hope to start using then.

IT-TechnologiesAdded, 2021-04-06:
As this issue is closed, can we expect a release with FreeCAD support soon?

Axel-G, 2021-04-06:
I'm still interested too :-)

OmniaGit, 2021-04-06:
We are working on it !!!

Axel-G, 2021-07-13:
Any news my Friend?

murych, 2021-07-15:
Would love to see FreeCAD integration too!

OmniaGit, 2021-07-22:
We are still working but FreeCad is quite a mess, because we have different version different operating system and also our ui is in pyside6 ..

Axel-G, 2022-01-14:
i m still seriously interested

mboscolo, 2022-01-16:
Dear Axel,

we are still an the way to the new client.

can you send me a direct mail to ***@***.***

regards,

Matteo

*OmniaSolutions*
/*Matteo Boscolo*
Technical Manager/
Cell. +39.327.1525045
Fax. +39.041.8623730
***@***.***
Skype: matteo.boscolo

OmniaGit, 2022-01-28:
Dear Axel,
we are ready for the first FreeCad testing version on odoo v15.

Let me know if you are still interested.

Regards,
Matteo

OmniaGit, 2022-09-10:
We are working hard to achieve the first public build of the freeCad integration.

[image]

stay tuned !!

Axel-G, 2022-09-11:
Hello Matteo,

do you have a testable version?

thx.

Axel-G, 2022-09-12:
sounds good.
with the freecad package you need plm box? because i use only open source, freecad on kde.

OmniaGit, 2022-09-19:
You do not need the plm box in order to work with odoo, just a new client application that we will publish very soon.
we are testing the best way to install it on freeCAD

I will take you update

danielliebherz, 2022-09-26:
i m interested too.

danielliebherz, 2022-09-30:
i m curios. i m useing a2plus workbench for assemblys and bom's. a connection to odoo would be great. maybe i can get it sponsored.

OmniaGit, 2022-09-30:
We are packaging the solution right now.
We target to go out with the first installation for the next week.
If you are coming at the odoo Experience 2022(BE) we will be there with a live demo.

danielliebherz, 2022-10-02:
Great. I am waiting patiently ;)

OmniaGit, 2022-10-04:
dear @danielliebherz 

could you give me the FreeCad version are you using and the os in witch FreeCad is running.
I have ready a Wind11 Freecad 0.20 installation ready to test

danielliebherz, 2022-10-05:
@OmniaGit

"unfortunately" i don't use windows.
fedora package, version 0.20.1

danielliebherz, 2022-11-23:
@OmniaGit  some news ?

variandp, 2024-01-22:
@OmniaGit @mboscolo 
Hi, i'm a student of industrial engineering in pancasila university, Jakarta. 
I have a strong desire to know or learn about how to integrating freecad and odoo, is it possible? 
I've been looking all this time for a way to make it possible, let's say just to automate some part on freecad so when i go back to odoo, it has replaced itself, i mean i'm sorry for my bad english, but you know what i'm trying to tell aight sir?
really hope for a reply from you sir, regards

-varian

variandp, 2024-01-22:
@OmniaGit @mboscolo
Hi, i'm a student of industrial engineering in pancasila university, Jakarta.
I have a strong desire to know or learn about how to integrating freecad and odoo, is it possible?
I've been looking all this time for a way to make it possible, let's say just to automate bill of material part on freecad so when i go back to odoo, it has replaced itself, i mean i'm sorry for my bad english, but you know what i'm trying to tell aight sir?
really hope for a reply from you sir, regards

-varian

GabrielVictorGVS, 2025-01-25:
@OmniaGit, Hello There! Do you have any news on this connector? It would be nice to have a download link for the alpha package or something similar, we're integrating odoo at my company.

OmniaGit, 2025-01-27:
The connection for the moment is in hold, as we are rewriting completely the client application.
We have in timeline for the end of February the delivery of the new client application at least for the Lite version.

we will keep you update once something will be also  be available for freecad.