OmniaGit/odooplm #33: is_plm to trigger backup - should we be more clear on UI?

duplomatic-anava, 2019-12-11:
@OmniaGit correct me if I am wrong; is_plm triggers only the backup of the file. What about adding to the user interface some help text for the user to understand that if this box is unchecked not backup is taken?

Let me know, I can handle this if you agree on the edit.

Alessio

OmniaGit, 2019-12-12:
Il plm flag indicate that a document is made from plm module or came from the client application, in those case we ally the back up rule.

It's ok for my side to also have a image or icon that indicate that the document is a plm one.
but I would live that flag read_only for the normal user.
what do you think ?

regards
Matteo

duplomatic-anava, 2019-12-12:
I agree with you that the normal user should not edit this field, only an administrator/moderator should be able to do so.

I would just add an help text telling the admin that this flag enables the backup of the file since it came from the PLM module or from the CAD clients. There could be a scenario where a lot of draft documents should not be backed up.

Do you agree to add the help text?

Alessio

OmniaGit, 2019-12-12:
yes please add it
regards,
Matteo

PR #41 (2020-01-14): help text for is_plm field
Closes #33
closed 2019-12-12 by OmniaGit