OmniaGit/odooplm #34: Document - name is editable for release document

duplomatic-anava, 2019-12-11:
The name for a released document is editable, but upon save, the name is correctly not changed. We should make it read-only.

In addition, the field "name" is present twice, also on the left to the revision. What about changing this to revision label so that we have only one name field?

I can take care of this if you agree.

Alessio

OmniaGit, 2019-12-12:
> The name for a released document is editable, but upon save, the name is correctly not changed. We should make it read-only.
sure you are right ..
> 
> In addition, the field "name" is present twice, also on the left to the revision. What about changing this to revision label so that we have only one name field?
yes please fix it, we did' ent notice, olse the double field in the same form can make strange behaviour on odoo
> 
> I can take care of this if you agree.
> 
> Alessio

duplomatic-anava, 2019-12-12:
I confirm that I can take care of this. I will open a pull request gathering the other fixes for the document part.

Alessio

closed 2019-12-12 by OmniaGit