OmniaGit/odooplm #35: Document - record can be saved even though no file has been added

duplomatic-anava, 2019-12-11:
A document record can be saved even if no file has been selected. In my opinion, we should not be able to save if a file is not present; upon creation, revision, always.

If needed, I can solve the issue with a pull request.

Thank you in advance.

Alessio

OmniaGit, 2019-12-12:
it's correct in this way, most of the time in the migratin to odooPLM from anothe plm we strognly use this 

> feature

regards
Matteo
 ..

closed 2019-12-12 by OmniaGit

duplomatic-anava, 2019-12-12:
Alright Matteo, thank you for the hint. I was just referring to the web interface, not making it a required database field.

It may make sense to create the document name and then upload the file later on the web interface, so we can leave it as it is right now.