OmniaGit/odooplm #39: Implementation ideas - Document categories

duplomatic-anava, 2019-12-12:
Dear @OmniaGit ,

internally, we are planning to create document categories following the scheme for product categories. Is this something you would like to see merged in this repo too?

So that all documents belonging to a certain category (quality, manufacturing, etc.) can be filtered.

Let me know.

Alessio

OmniaGit, 2019-12-12:
Thanks,
but  for the moment we do not have any request in this terms.
also saving file from the cad application will impact on this too, so we prefer to leave it as it is

regards,
Matteo

closed 2019-12-12 by OmniaGit

duplomatic-anava, 2019-12-12:
Alright Matteo, I will leave a branch in our public fork in case you changed your mind in the future :)

Alessio

OmniaGit, 2019-12-12:
Thanks any way ..
regards,
Matteo