OmniaGit/odooplm #47: [13.0] plm_document_multi_side invalid field name

InforMedic, 2020-01-20:
Hello,

again on my odoo.sh instance installing the module plm_document_multi_side gives me the following error:

<details>
  <summary>Click to expand traceback!</summary>
<p>

```
2020-01-20 14:16:04,415 7 ERROR odoo-v13-development-821818 odoo.modules.registry: Failed to load registry
Traceback (most recent call last):
File "/home/odoo/src/odoo/odoo/tools/convert.py", line 713, in parse
self._tag_root(de)
File "/home/odoo/src/odoo/odoo/tools/convert.py", line 675, in _tag_root
f(rec)
File "/home/odoo/src/odoo/odoo/tools/convert.py", line 675, in _tag_root
f(rec)
File "/home/odoo/src/odoo/odoo/tools/convert.py", line 578, in _tag_record
record = model._load_records([data], self.mode == 'update')
File "/home/odoo/src/odoo/odoo/models.py", line 4070, in _load_records
records = self._load_records_create([data['values'] for data in to_create])
File "/home/odoo/src/odoo/odoo/models.py", line 3984, in _load_records_create
return self.create(values)
File "<decorator-gen-40>", line 2, in create
File "/home/odoo/src/odoo/odoo/api.py", line 325, in _model_create_single
return self.browse().concat(*(create(self, vals) for vals in arg))
File "/home/odoo/src/odoo/odoo/api.py", line 325, in <genexpr>
return self.browse().concat(*(create(self, vals) for vals in arg))
File "/home/odoo/src/odoo/odoo/addons/base/models/ir_cron.py", line 71, in create
return super(ir_cron, self).create(values)
File "<decorator-gen-3>", line 2, in create
File "/home/odoo/src/odoo/odoo/api.py", line 343, in _model_create_multi
return create(self, [arg])
File "/home/odoo/src/odoo/odoo/models.py", line 3689, in create
raise ValueError("Invalid field %r on model %r" % (key, self._name))
ValueError: Invalid field 'function' on model 'ir.cron'
odoo.tools.convert.ParseError: "Invalid field 'function' on model 'ir.cron'" while parsing /home/odoo/src/user/plm_document_multi_site/data/server_action.xml:2, near
<odoo>
<data noupdate="1">

<record id="ir_cron_syncronize" model="ir.cron">
<field name="name">Syncronize</field>
<field name="active" eval="False"/>
<field name="interval_number">5</field>
<field name="interval_type">minutes</field>
<field name="numbercall">-1</field>
<field name="doall" eval="False"/>
<field name="model_id" ref="model_plm_document_action_syncronize"/>
<field name="function">syncronize</field>
<field name="args">()</field>
</record>
</data>

<data>
<record id="action_syncronize" model="ir.actions.server">
<field name="name">Synconize document</field>
<field name="model_id" ref="model_plm_document_action_syncronize"/>
<field name="code">env['plm.document.action.syncronize'].syncronize()</field>
<field name="state">code</field>
<field name="type">ir.actions.server</field>
</record>

<menuitem id="menu_syncronize_scheduler" name="Run Document Syncronize " action="action_syncronize" groups="plm.group_plm_admin" parent="plm.plm_menu_dictionaries"/>
</data>
</odoo>
2020-01-20 14:16:04,425 7 INFO odoo-v13-development-821818 odoo.service.server: Initiating shutdown
2020-01-20 14:16:04,425 7 INFO odoo-v13-development-821818 odoo.service.server: Hit CTRL-C again or send a second signal to force the shutdown.
odoo-bin process returned error code 255.
```
</p>
</details>

I have no idea if this is maybe a protected field name string?!? no clue - can you have a look at this?

Many thx!

vipin-sigb-dev, 2020-01-21:
<record id="ir_cron_syncronize" model="ir.cron">
<field name="name">Syncronize</field>
<field name="active" eval="False"/>
<field name="interval_number">5</field>
<field name="interval_type">minutes</field>
<field name="numbercall">-1</field>
<field name="doall" eval="False"/>
<field name="model_id" ref="model_plm_document_action_syncronize"/>
<field name="code">model.syncronize()</field>
</record>

HI, please use this code.seems to be  ir_cron has Invalid field 'function" in odoo13.

OmniaGit, 2020-01-22:
Thanks for the bug report and the solution..
you can find the fix with the following commit:
https://github.com/OmniaGit/odooplm/commit/c8023c846c65d85397fc848e9a919969cbe14875

regards

closed 2020-01-22 by OmniaGit