OmniaGit/odooplm #48: [12.0] website and plm module conflict

duplomatic-anava, 2020-01-24:
@OmniaGit I may be experiencing a conflict between odooplm and the website module upon installation.

First, I am installing the plm module, and then the website one. The error is as follows:
`
ValueError: Element '<field name="mimetype">' cannot be located in parent view
`

Can you please confirm it? I can take care of that later on next week if confirmed.

OmniaGit, 2020-01-25:
Dear Alessio,
I'm not able to reproduce the error.
i just test it on odooSH
https://odooplm-12-0-833536.dev.odoo.com/abc?

and the module work like a charm.
did you install any other modules ?
regards,
Matteo

duplomatic-anava, 2020-01-27:
Dear Matteo,

we must have broken something at some point, but with latest odoo (today) and 12.0 of odooplm there are no issues.

Thanks!

Alessio

closed 2020-01-27 by duplomatic-anava