OmniaGit/odooplm #50: odoo

anviz-github, 2020-03-06:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

OmniaGit, 2020-03-06:
Dear Anviz,
We check several time the issue you report but we are not able to reproduce it.

The algorithm that is in charged of adding part in the bom eco make a compare only for the following attributes:

>   (line.product_uom_id, line.product_qty, line.operation_id) != (old_line.product_uom_id, old_line.product_qty, old_line.operation_id)

Please have a check of all the bom involved to see if there is any difference.
Be aware that odoo make a new bom for each eco, and put the new one in active=False mode.

In order to investigate more we need to have a base model that help us to reproduce the behaviour

regards

anviz-github, 2020-03-07:
https://youtu.be/KDEVUT7XQ4g
Thank you！

OmniaGit, 2020-03-08:
Dear Anviz,
thanks for your reporting, we are able to reproduce the problem, and we fix it with the following commit
https://github.com/OmniaGit/odooplm/commit/d4445b18a7375dfdeb2fdc9371337c152f67d6d9

thanks in advance,
regards

closed 2020-03-08 by OmniaGit