OmniaGit/odooplm #60: Issue doing inventory adjustment using import

sandertu, 2021-05-24:
Happens even when importing an exported inventory adjustment.
Using Odoo 13 and odooplm branch 13.00

Traceback (most recent call last):
  File "D:\Odoo 13.0\Odoo 13\server\odoo\http.py", line 624, in _handle_exception
    return super(JsonRequest, self)._handle_exception(exception)
  File "D:\Odoo 13.0\Odoo 13\server\odoo\http.py", line 310, in _handle_exception
    raise pycompat.reraise(type(exception), exception, sys.exc_info()[2])
  File "D:\Odoo 13.0\Odoo 13\server\odoo\tools\pycompat.py", line 14, in reraise
    raise value
  File "D:\Odoo 13.0\Odoo 13\server\odoo\http.py", line 669, in dispatch
    result = self._call_function(**self.params)
  File "D:\Odoo 13.0\Odoo 13\server\odoo\http.py", line 350, in _call_function
    return checked_call(self.db, *args, **kwargs)
  File "D:\Odoo 13.0\Odoo 13\server\odoo\service\model.py", line 94, in wrapper
    return f(dbname, *args, **kwargs)
  File "D:\Odoo 13.0\Odoo 13\server\odoo\http.py", line 339, in checked_call
    result = self.endpoint(*a, **kw)
  File "D:\Odoo 13.0\Odoo 13\server\odoo\http.py", line 915, in __call__
    return self.method(*args, **kw)
  File "D:\Odoo 13.0\Odoo 13\server\odoo\http.py", line 515, in response_wrap
    response = f(*args, **kw)
  File "D:\Odoo 13.0\Odoo 13\server\odoo\addons\web\controllers\main.py", line 1339, in call_kw
    return self._call_kw(model, method, args, kwargs)
  File "D:\Odoo 13.0\Odoo 13\server\odoo\addons\web\controllers\main.py", line 1331, in _call_kw
    return call_kw(request.env[model], method, args, kwargs)
  File "D:\Odoo 13.0\Odoo 13\server\odoo\api.py", line 387, in call_kw
    result = _call_kw_multi(method, model, args, kwargs)
  File "D:\Odoo 13.0\Odoo 13\server\odoo\api.py", line 374, in _call_kw_multi
    result = method(recs, *args, **kwargs)
  File "D:\Odoo 13.0\Odoo 13\server\odoo\addons\base_import\models\base_import.py", line 924, in do
    import_result = model.load(import_fields, data)
  File "D:\Odoo 13.0\Odoo 13\server\odoo\models.py", line 1043, in load
    for id, xid, record, info in converted:
  File "D:\Odoo 13.0\Odoo 13\server\odoo\models.py", line 1193, in _convert_records
    converted = convert(record, functools.partial(_log, extras, stream.index))
  File "D:\Odoo 13.0\Odoo 13\server\odoo\addons\base\models\ir_fields.py", line 80, in fn
    converted[field], ws = converters[field](value)
  File "D:\Odoo 13.0\Odoo 13\server\odoo\addons\base\models\ir_fields.py", line 416, in _str_to_many2one
    id, _, w2 = self.db_id_for(model, field, subfield, record[subfield])
  File "D:\Odoo 13.0\Odoo 13\server\odoo\addons\base\models\ir_fields.py", line 353, in db_id_for
    ids = RelatedModel.name_search(name=value, operator='=')
  File "D:\Odoo 13.0\Odoo 13\server\odoo\models.py", line 1740, in name_search
    return self._name_search(name, args, operator, limit=limit)
  File "D:\Odoo 13.0\Odoo 13\server\odoo\addons\plm\models\product_product.py", line 1791, in _name_search
TypeError: can only concatenate list (not "NoneType") to list

sandertu, 2021-05-24:
Imported test file contains

<meta name="ProgId" content="Excel.Sheet">
<meta name="Generator" content="Microsoft Excel 15">
<link id="Main-File" rel="Main-File" href="file:///C:/Users/kasutaja/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel="File-List" href="file:///C:/Users/kasutaja/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">

Product
--
[13753457]   123
[13753457]   123

OmniaGit, 2021-05-24:
Dear sandertu,

Thank you for the reporting.
I made a commit to fix this issue, please update your plm modules using GIT.
At your disposal.

Best Regards,
Daniel

closed 2021-05-25 by OmniaGit