OmniaGit/odooplm #67: Part image in Odoo 14 not updated after part is modified

etoimene, 2022-04-13:
**Describe the bug** A clear and concise description of what the bug is.
We change part geometry in SolidWorks 2022 and after save in Odoo (tried check-in too) part image on LinkedDocuments tab is not updated.

**To Reproduce** Steps to reproduce the behavior:

1. We change part geometry in SolidWorks 2022
2. Using PLM Save button we save it to Odoo

**Expected behavior** A clear and concise description of what you expected to happen.
LinkedDocuments image should be like new geometry.

**Screenshots** If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**

- OS: Win10
- Browser : any
- Version [e.g. 22]

**Additional context** Add any other context about the problem here.

OmniaGit, 2022-04-19:
Dear,

did you save locally(solidworks) before save into add-in application ?
there is also an option on the client that make the save operation for you, does this setting is enabled ?

in order to further investigate on your behaviour we need to know the client version that are you using (complete name of the installer)

thanks for your report

etoimene, 2022-04-19:
Thanks for your great product. :)

We have explicitly saved part using SolidWorks before attempting PLM Save.
Not sure which setting is needed for this automated save operation.
I will update PLM client version info when I get access to it, but downloaded from sourceforge in the last 40-50 days, so fairly recent.

etoimene, 2022-04-19:
[image]
Here is the version.

OmniaGit, 2022-04-20:
Update  your odooplm modules from github on odoo server performing un upgrade on the plm module
download the latest client at this link https://sourceforge.net/projects/openerpplm/files/Client/Candidate/OdooPlm_2019_odoov13_31032022_095826.exe/download
the system should work out of the box.

let me know if it fix the problem
regards

etoimene, 2022-04-23:
For some parts it works, for some not. 
With newly added parts, there are no problems (as far as I can see).

We installed the client you suggested and have Odoo modules.
"Module Name","Technical Name","Latest Version","Status"
"Product Lifecycle Management","plm","14.0.2","Installed"
"PLM Automate Normal Bom","plm_automate_normal_bom","14.0.1.0.0","Installed"
"PLM Compare Bom","plm_compare_bom","14.0.1.0.0","Installed"
"PLM Engineering","plm_engineering","14.0.1.0.0","Installed"

etoimene, 2022-04-29:
That is set. Without that, it is impossible to save the part in first place.

Can you suggest what should I look for in eg. application.log?

It works on some parts and not on other, so my process is OK, but I think some properties are different.

etoimene, 2022-04-29:
Part is saved correctly, but image is missing. Maybe image name has some strange characters or it is attempted to be stored somewhere on the client where it shouldn't be ...

OmniaGit, 2022-04-29:
for the image you need to set properly windows explorer in order to upload the preview.

[image]
 
regards

etoimene, 2022-04-29:
Oooh, great! That makes sense. 
It always uploads the latest thumbnail?

OmniaGit, 2022-05-02:
Yes, during the saving procedure CAD-> odoo we extract the preview from the files and upload it
regards

etoimene, 2022-05-06:
I tried again and with new parts it works as expected.

Parts that are already in the Odoo without those images do not get new images, despite correct thumbnail update on Windows file explorer.

OmniaGit, 2022-05-07:
the file that has no change is not updated.
the system in order to be very smart upload only the files that are changed.

regards,
Matteo

etoimene, 2022-05-08:
From what I can see, Windows explorer thumbnail was updated, but not DB. We did not change part name (engineering code).
I'll give it another test.

What do you consider change?

OmniaGit, 2022-05-09:
it means that the file date is changed, than the client application will upload the new image to odoo using the save button

regards,
Matteo

etoimene, 2022-05-09:
Thanks! Somehow I thought timestamp is used.
I will make another round of tests. Maybe on the different PC.

etoimene, 2022-05-23:
Some new info.
We have 2 things that could influence this.
1. odoo server is not synced with ntp
2. Thumbnails on this machine are sometimes changed, sometimes not

I'll investigate further and post result here.

etoimene, 2022-06-13:
After ntp sync issues on one machine are gone.
Other one is still bugging us. That one beheves strangely in general, so might be some Windows issue.

closed 2022-06-13 by OmniaGit