OmniaGit/odooplm #68: Connector for AutoCAD

FernandoJFernandez, 2022-04-29:
Hi,
Can anyone confirm if the Client version OdooPlm_2019_odoov13_31032022_095826.exe is supported with Autodesk Autocad 2021?
I'm trying to install the client connector for autocad but the menu does not appear.
I'm trying to load the autocad folder of the connector inside autocad 2021, but I don't see any result, I don't know if I'm doing something wrong.

[image]

I would appreciate some kind of documentation or any help.

Thank you.

OmniaGit, 2022-04-29:
Dear Fernando,
we test the solutions on autocad 2022 and it works out of the box.

the client application use dll-addin in order to create the toolbar.

in order to be sure that the installation work follow the current step.

- disable the UAC
- be machine administrator
- install the software as administrator

if the toolbar dose not appear us the autocad command  netload and load the OdooPLMAutocad.dll  under C:\Program Files\Odoo\Integrations\Autocad
(https://knowledge.autodesk.com/support/autocad/learn-explore/caas/sfdcarticles/sfdcarticles/How-to-autoload-DLLs-with-AutoCAD.html)
should also be possible that the first time the toolbar is hidden so you need to unhide

let me know if it works 
regards

FernandoJFernandez, 2022-04-29:
It's work fine. Thank you very much.

closed 2022-04-29 by FernandoJFernandez

OmniaGit, 2022-05-02:
Perfect..