OmniaGit/odooplm #69: Error appear "Missing engineering code for plm data" when save Engineering Parts

bi7jta, 2022-05-08:
**Describe the bug** A clear and concise description of what the bug is.
Hi to all good man,
I think anyone have experience deploy this PLM module can help me , I have got a error when I create and save "[Engineering Parts]"

Thank you very much !

Error: "Missing engineering code for plm data"

Request URL: http://192.168.2.15:8069/web/dataset/call_kw/product.product/create
{
    "id": 626,
    "jsonrpc": "2.0",
    "method": "call",
    "params": {
        "args": [
            {
                "service_type": "manual",
                "attribute_line_ids": [],
                "type": "consu",
                "is_published": false,
                "image_1920": false,
                "__last_update": "2022-05-08 01:28:03",
                "name": "Demo",
                "sale_ok": false,
                "purchase_ok": true,
                "std_description": false,
                "readonly_std_umc1": false,
                "readonly_std_umc2": false,
                "readonly_std_umc3": false,
                "std_umc1": false,
                "std_value1": 0,
                "std_umc2": false,
                "std_value2": 0,
                "std_umc3": false,
                "std_value3": 0,
                "source_product": false,
                "desc_modify": "",
                "active": true,
                "detailed_type": "consu",
                "invoice_policy": "order",
                "expense_policy": "no",
                "uom_id": 1,
                "base_unit_count": 1,
                "base_unit_id": false,
                "uom_po_id": 1,
                "list_price": 1,
                "lst_price": 1,
                "taxes_id": [
                    [
                        6,
                        false,
                        [
                            20
                        ]
                    ]
                ],
                "standard_price": 0,
                "default_code": false,
                "barcode": false,
                "categ_id": 1,
                "company_id": false,
                "description": "<p>.</p>",
                "accessory_product_ids": [
                    [
                        6,
                        false,
                        []
                    ]
                ],
                "alternative_product_ids": [
[... 114 lines cut ...]
        "data": {
            "name": "odoo.exceptions.UserError",
            "debug": "Traceback (most recent call last):\n  File \"C:\\Program Files\\Odoo 15.0.20220426\\server\\odoo\\addons\\base\\models\\ir_http.py\", line 237, in _dispatch\n    result = request.dispatch()\n  File \"C:\\Program Files\\Odoo 15.0.20220426\\server\\odoo\\http.py\", line 688, in dispatch\n    result = self._call_function(**self.params)\n  File \"C:\\Program Files\\Odoo 15.0.20220426\\server\\odoo\\http.py\", line 360, in _call_function\n    return checked_call(self.db, *args, **kwargs)\n  File \"C:\\Program Files\\Odoo 15.0.20220426\\server\\odoo\\service\\model.py\", line 94, in wrapper\n    return f(dbname, *args, **kwargs)\n  File \"C:\\Program Files\\Odoo 15.0.20220426\\server\\odoo\\http.py\", line 349, in checked_call\n    result = self.endpoint(*a, **kw)\n  File \"C:\\Program Files\\Odoo 15.0.20220426\\server\\odoo\\http.py\", line 917, in __call__\n    return self.method(*args, **kw)\n  File \"C:\\Program Files\\Odoo 15.0.20220426\\server\\odoo\\http.py\", line 536, in response_wrap\n    response = f(*args, **kw)\n  File \"C:\\Program Files\\Odoo 15.0.20220426\\server\\odoo\\addons\\web\\controllers\\main.py\", line 1348, in call_kw\n    return self._call_kw(model, method, args, kwargs)\n  File \"C:\\Program Files\\Odoo 15.0.20220426\\server\\odoo\\addons\\web\\controllers\\main.py\", line 1340, in _call_kw\n    return call_kw(request.env[model], method, args, kwargs)\n  File \"C:\\Program Files\\Odoo 15.0.20220426\\server\\odoo\\api.py\", line 462, in call_kw\n    result = _call_kw_model_create(method, model, args, kwargs)\n  File \"C:\\Program Files\\Odoo 15.0.20220426\\server\\odoo\\api.py\", line 442, in _call_kw_model_create\n    result = method(recs, *args, **kwargs)\n  File \"<decorator-gen-286>\", line 2, in create\n  File \"C:\\Program Files\\Odoo 15.0.20220426\\server\\odoo\\api.py\", line 391, in _model_create_single\n    return create(self, arg)\n  File \"C:\\Program Files\\Odoo 15.0.20220426\\server\\odoo\\addons\\plm\\models\\product_product.py\", line 1002, in create\n    raise ex\n  File \"C:\\Program Files\\Odoo 15.0.20220426\\server\\odoo\\addons\\plm\\models\\product_product.py\", line 998, in create\n    res = super(PlmComponent, self).create(vals)\n  File \"<decorator-gen-166>\", line 2, in create\n  File \"C:\\Program Files\\Odoo 15.0.20220426\\server\\odoo\\api.py\", line 412, in _model_create_multi\n    return create(self, [arg])\n  File \"C:\\Program Files\\Odoo 15.0.20220426\\server\\odoo\\addons\\product\\models\\product.py\", line 362, in create\n    products = super(ProductProduct, self.with_context(create_product_product=True)).create(vals_list)\n  File \"<decorator-gen-130>\", line 2, in create\n  File \"C:\\Program Files\\Odoo 15.0.20220426\\server\\odoo\\api.py\", line 413, in _model_create_multi\n    return create(self, arg)\n  File \"C:\\Program Files\\Odoo 15.0.20220426\\server\\odoo\\addons\\mail\\models\\mail_thread.py\", line 265, in create\n    threads = super(MailThread, self).create(vals_list)\n  File \"<decorator-gen-67>\", line 2, in create\n  File \"C:\\Program Files\\Odoo 15.0.20220426\\server\\odoo\\api.py\", line 413, in _model_create_multi\n    return create(self, arg)\n  File \"C:\\Program Files\\Odoo 15.0.20220426\\server\\odoo\\addons\\base\\models\\ir_fields.py\", line 613, in create\n    recs = super().create(vals_list)\n  File \"<decorator-gen-13>\", line 2, in create\n  File \"C:\\Program Files\\Odoo 15.0.20220426\\server\\odoo\\api.py\", line 413, in _model_create_multi\n    return create(self, arg)\n  File \"C:\\Program Files\\Odoo 15.0.20220426\\server\\odoo\\models.py\", line 4103, in create\n    fields[0].determine_inverse(batch_recs)\n  File \"C:\\Program Files\\Odoo 15.0.20220426\\server\\odoo\\fields.py\", line 1275, in determine_inverse\n    getattr(records, self.inverse)()\n  File \"C:\\Program Files\\Odoo 15.0.20220426\\server\\odoo\\addons\\product\\models\\product.py\", line 185, in _set_image_1920\n    return self._set_template_field('image_1920', 'image_variant_1920')\n  File \"C:\\Program Files\\Odoo 15.0.20220426\\server\\odoo\\addons\\product\\models\\product.py\", line 174, in _set_template_field\n    record[variant_field] = False\n  File \"C:\\Program Files\\Odoo 15.0.20220426\\server\\odoo\\models.py\", line 5894, in __setitem__\n    return self._fields[key].__set__(self, value)\n  File \"C:\\Program Files\\Odoo 15.0.20220426\\server\\odoo\\fields.py\", line 1217, in __set__\n    records.write({self.name: write_value})\n  File \"C:\\Program Files\\Odoo 15.0.20220426\\server\\odoo\\addons\\plm\\models\\product_product.py\", line 1035, in write\n    self.checkFromOdooPlm()\n  File \"C:\\Program Files\\Odoo 15.0.20220426\\server\\odoo\\addons\\plm\\models\\product_product.py\", line 1131, in checkFromOdooPlm\n    raise UserError(\"Missing engineering code for plm data\")\nException\n\nThe above exception was the direct cause of the following exception:\n\nTraceback (most recent call last):\n  File \"C:\\Program Files\\Odoo 15.0.20220426\\server\\odoo\\http.py\", line 644, in _handle_exception\n    return super(JsonRequest, self)._handle_exception(exception)\n  File \"C:\\Program Files\\Odoo 15.0.20220426\\server\\odoo\\http.py\", line 302, in _handle_exception\n    raise exception.with_traceback(None) from new_cause\nodoo.exceptions.UserError: Missing engineering code for plm data\n",
            "message": "Missing engineering code for plm data",
            "arguments": [
                "Missing engineering code for plm data"
            ],
            "context": {}
        }
    }
}

**Expected behavior** A clear and concise description of what you expected to happen.

**Screenshots** If applicable, add screenshots to help explain your problem.

[](url)

**Desktop (please complete the following information):**

- OS: [e.g. iOS]
- Browser [e.g. chrome, safari]
- Version [e.g. 22]

**Additional context** Add any other context about the problem here.

OmniaGit, 2022-05-09:
Dear,
you need to add also the engineering_code field to the create function.

regards,
Matteo

bi7jta, 2022-05-09:
> engineering_code

OK, thank you very much, I will try . Best regards!

alenhsu038, 2022-05-12:
Dear Sir, 
I am new comer to use odooplm. I encountered the same problem" Missing engineering code for plm data". 
May I have detailed procedure about how  to "add the engineering_code field to the create function."

My Odoo is 15.0-20220501 (Community Edition), Thanks.

Thanks.

OmniaGit, 2022-05-17:
self.env['product.product'].crete({'engennering_code': 'x102030',....}

should fix your problem

regards

closed 2022-06-22 by OmniaGit