OmniaGit/odooplm #70: Error when saving product with multiple attribute values. v15

j540, 2022-05-14:
I am expereincing an error when saving a product with multiple attribute values. 

Odoo v15 Enterprise

Steps to reproduce the behavior:
1. Go to Inventory > Products > Create
2. Click Attributes & Values
3. Add an attribute. (i.e. Color), and multiple values (i.e. Black, White)
4. Click Save

The following error message is presented:

RPC_ERROR
Odoo Server Error
Traceback (most recent call last):
  File "/home/odoo/src/odoo/odoo/addons/base/models/ir_http.py", line 237, in _dispatch
    result = request.dispatch()
  File "/home/odoo/src/odoo/odoo/http.py", line 686, in dispatch
    result = self._call_function(**self.params)
  File "/home/odoo/src/odoo/odoo/http.py", line 358, in _call_function
    return checked_call(self.db, *args, **kwargs)
  File "/home/odoo/src/odoo/odoo/service/model.py", line 94, in wrapper
    return f(dbname, *args, **kwargs)
  File "/home/odoo/src/odoo/odoo/http.py", line 347, in checked_call
    result = self.endpoint(*a, **kw)
  File "/home/odoo/src/odoo/odoo/http.py", line 915, in __call__
    return self.method(*args, **kw)
  File "/home/odoo/src/odoo/odoo/http.py", line 534, in response_wrap
    response = f(*args, **kw)
  File "/home/odoo/src/odoo/addons/web/controllers/main.py", line 1348, in call_kw
    return self._call_kw(model, method, args, kwargs)
  File "/home/odoo/src/odoo/addons/web/controllers/main.py", line 1340, in _call_kw
    return call_kw(request.env[model], method, args, kwargs)
  File "/home/odoo/src/odoo/odoo/api.py", line 462, in call_kw
    result = _call_kw_model_create(method, model, args, kwargs)
  File "/home/odoo/src/odoo/odoo/api.py", line 442, in _call_kw_model_create
    result = method(recs, *args, **kwargs)
  File "<decorator-gen-288>", line 2, in create
  File "/home/odoo/src/odoo/odoo/api.py", line 391, in _model_create_single
    return create(self, arg)
  File "/home/odoo/src/user/plm_auto_internalref/models/product_template.py", line 46, in create
    return super(ProductTemplateExtension, self).create(vals)
  File "<decorator-gen-271>", line 2, in create
  File "/home/odoo/src/odoo/odoo/api.py", line 391, in _model_create_single
    return create(self, arg)
  File "/home/odoo/src/user/plm/models/product_template.py", line 189, in create
    return super(ProductTemplateExtension, self).create(vals)
  File "<decorator-gen-164>", line 2, in create
  File "/home/odoo/src/odoo/odoo/api.py", line 412, in _model_create_multi
    return create(self, [arg])
  File "/home/odoo/src/odoo/addons/product/models/product_template.py", line 473, in create
    templates._create_variant_ids()
  File "/home/odoo/src/odoo/addons/product/models/product_template.py", line 730, in _create_variant_ids
    variants_to_activate.write({'active': True})
  File "/home/odoo/src/user/plm_product_description_language_helper/models/product_product.py", line 104, in write
    return super(ProductProductExtension, self).write(vals)
  File "/home/odoo/src/user/plm_auto_internalref/models/product_product.py", line 54, in write
    res = super(ProductProductExtension, self).write(vals)
  File "/home/odoo/src/user/plm/models/product_product.py", line 1034, in write
    if self._origin.id:
  File "/home/odoo/src/odoo/odoo/fields.py", line 4061, in __get__
    raise ValueError("Expected singleton: %s" % record)
Exception

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/odoo/src/odoo/odoo/http.py", line 642, in _handle_exception
    return super(JsonRequest, self)._handle_exception(exception)
  File "/home/odoo/src/odoo/odoo/http.py", line 300, in _handle_exception
    raise exception.with_traceback(None) from new_cause
ValueError: Expected singleton: product.product(2620, 2621)

j540, 2022-05-14:
Uninstalling module: plm_auto_internalref does not fix issue.

j540, 2022-05-14:
Uninstalling odooplm app fixes the issue.

OmniaGit, 2022-05-16:
Dear,

you are reporting a bug that is already fixed in branch https://github.com/OmniaGit/odooplm/tree/15.0_issue_improve that will be merged very soon.

let me know if it fix
regards

closed 2022-05-17 by j540