OmniaGit/odooplm #71: 'Connection aborted.', error(10054, 'An existing connection was forcibly closed by the remote host')

j540, 2022-06-19:
When saving a solidworks assembly from the OdooPLM client, an intermittent error appears. And the save fails.

'Connection aborted.', error(10054, 'An existing connection was forcibly closed by the remote host')

By checking the OdooPlm 'DB Thread Action' page, this error can be see, and the specific solidworks part that is causing the error can be seen. The error does not constantly appear on the same part. 

OdooPlm v 15.0
Solidworks 2022
OdooPlm Client: OdooPlm_2019_odoov13_16062022_180028.exe

Solidworks Assembly Size: About 100 parts. 

Thank you for any assistance.

j540, 2022-06-19:
Attached is the output of the DB Thread page. 

[image]

j540, 2022-06-20:
Also, the odoo instance is deployed on odoo.sh

j540, 2022-06-22:
@OmniaGit , is there an official 'Large Assembly' upload strategy?

OmniaGit, 2022-06-22:
Dear @j540 

how your assembly is made ?
do you have only standard solidworks part or are you using some slt .. stp file ?

the client application upload the file in a thread way and upload a pull of 4 file per time.
and you can use this option in order to control the number of simultaneous process to use
 
[image]

we do not limit the upload dimension of the file in server controller.

a good strategy could be to save the sub assembly and check-in it after the save operation is completed, so the system dose not upload the sub assembly files.

let me know if this let  you solve the problem.

regards,
Matteo

OmniaGit, 2022-06-22:
> Attached is the output of the DB Thread page.
> 
> [image]

from your image it seems that the Motor gasket is the problem . try to unload it in order to understand if the problem is there..

j540, 2022-06-26:
Thank you very much, for the help. I will try your suggestions.

OmniaGit, 2022-06-27:
Let  us know if you solve the problem
regards

closed 2022-06-27 by j540

OmniaGit, 2022-06-27:
Thanks for the feedback
regards