OmniaGit/odooplm #76: Additional documentation on PLM clinet options

etoimene, 2023-11-10:
Dear Omnia team!

I'd like to report that we have succesfully tested v16 version integration with SolidWorks. 
It took me a while to realize that 
latest version https://sourceforge.net/projects/openerpplm/files/latest/download
and
latest stable https://sourceforge.net/projects/openerpplm/files/Client/Stable/
do not work with the latest v16 server software.

I used the latest candidate client https://sourceforge.net/projects/openerpplm/files/Client/Candidate/
and all worked well. :)

Suggestion #1: Please make some readme about prerequisites! 

Suggestion #2:
Since client is not an open source, please describe better available options.
https://odooplm.omniasolutions.website/login#Options

This way it is more like hit and miss. :(

Also, we noticed that product.template name is not populated from client side data like before (from part properties). Now we have to enter Description manually.
This was intentional change or ...?

Thanks!

Kind regards!

OmniaGit, 2023-11-10:
Dear etoimene,

Thanks for your kind e-mail,we will take care of your suggestion in order to improve our website.

Regarding the product.template problem odooplm has a property bridge that enable to map CAD property with odoo property.

We provide this kind of service in the SP package .

https://www.omniasolutions.website/shop/category/odooplm-starting-package-5

Regards,

etoimene, 2023-11-10:
Thanks for the quick reply. This SP is a bit above our needs and budget ATM. We'll work it out.

etoimene, 2023-11-10:
What is CustomProcedures mentioned within SP?

OmniaGit, 2023-11-10:
CustomProcedures is a custom py file that customize the client application upon customer needs

OmniaGit, 2023-11-10:
You are welcome ...

closed 2023-11-10 by OmniaGit