OmniaGit/odooplm #79: Error during installation

FalkoG123, 2023-11-16:
RPC_ERROR
Odoo Server Error

Traceback (most recent call last):
  File "/odoo17/odoo/odoo/http.py", line 1732, in _serve_db
    return service_model.retrying(self._serve_ir_http, self.env)
  File "/odoo17/odoo/odoo/service/model.py", line 133, in retrying
    result = func()
  File "/odoo17/odoo/odoo/http.py", line 1759, in _serve_ir_http
    response = self.dispatcher.dispatch(rule.endpoint, args)
  File "/odoo17/odoo/odoo/http.py", line 1960, in dispatch
    result = self.request.registry['ir.http']._dispatch(endpoint)
  File "/odoo17/odoo/addons/website/models/ir_http.py", line 235, in _dispatch
    response = super()._dispatch(endpoint)
  File "/odoo17/odoo/odoo/addons/base/models/ir_http.py", line 207, in _dispatch
    result = endpoint(**request.params)
  File "/odoo17/odoo/odoo/http.py", line 722, in route_wrapper
    result = endpoint(self, *args, **params_ok)
  File "/odoo17/odoo/addons/web/controllers/dataset.py", line 28, in call_button
    action = self._call_kw(model, method, args, kwargs)
  File "/odoo17/odoo/addons/web/controllers/dataset.py", line 20, in _call_kw
    return call_kw(request.env[model], method, args, kwargs)
  File "/odoo17/odoo/odoo/api.py", line 466, in call_kw
    result = _call_kw_multi(method, model, args, kwargs)
  File "/odoo17/odoo/odoo/api.py", line 453, in _call_kw_multi
    result = method(recs, *args, **kwargs)
  File "<decorator-gen-76>", line 2, in button_immediate_install
  File "/odoo17/odoo/odoo/addons/base/models/ir_module.py", line 75, in check_and_log
    return method(self, *args, **kwargs)
  File "/odoo17/odoo/odoo/addons/base/models/ir_module.py", line 459, in button_immediate_install
    return self._button_immediate_function(type(self).button_install)
  File "/odoo17/odoo/odoo/addons/base/models/ir_module.py", line 583, in _button_immediate_function
    registry = modules.registry.Registry.new(self._cr.dbname, update_module=True)
  File "<decorator-gen-16>", line 2, in new
  File "/odoo17/odoo/odoo/tools/func.py", line 87, in locked
    return func(inst, *args, **kwargs)
  File "/odoo17/odoo/odoo/modules/registry.py", line 113, in new
    odoo.modules.load_modules(registry, force_demo, status, update_module)
  File "/odoo17/odoo/odoo/modules/loading.py", line 480, in load_modules
    processed_modules += load_marked_modules(env, graph,
  File "/odoo17/odoo/odoo/modules/loading.py", line 364, in load_marked_modules
    loaded, processed = load_module_graph(
  File "/odoo17/odoo/odoo/modules/loading.py", line 205, in load_module_graph
    registry.init_models(env.cr, model_names, {'module': package.name}, new_install)
  File "/odoo17/odoo/odoo/modules/registry.py", line 582, in init_models
    func()
  File "/odoo17/odoo/odoo/addons/base/models/ir_model.py", line 1916, in _reflect_relation
    self.env.invalidate_all()
  File "/odoo17/odoo/odoo/api.py", line 702, in invalidate_all
    self.flush_all()
  File "/odoo17/odoo/odoo/api.py", line 712, in flush_all
    self._recompute_all()
  File "/odoo17/odoo/odoo/api.py", line 708, in _recompute_all
    self[field.model_name]._recompute_field(field)
  File "/odoo17/odoo/odoo/models.py", line 6825, in _recompute_field
    field.recompute(records)
  File "/odoo17/odoo/odoo/fields.py", line 1365, in recompute
    apply_except_missing(self.compute_value, recs)
  File "/odoo17/odoo/odoo/fields.py", line 1338, in apply_except_missing
    func(records)
  File "/odoo17/odoo/odoo/fields.py", line 1387, in compute_value
    records._compute_field_value(self)
  File "/odoo17/odoo/addons/mail/models/mail_thread.py", line 424, in _compute_field_value
    return super()._compute_field_value(field)
  File "/odoo17/odoo/odoo/models.py", line 4848, in _compute_field_value
    fields.determine(field.compute, self)
  File "/odoo17/odoo/odoo/fields.py", line 101, in determine
    return needle(*args)
  File "/odoo17/custom/addons/odooplm/plm/models/ir_attachment.py", line 1305, in _checkSavingError
    ir_attachment_id.has_error = not ir_attachment_id.is_last_save_ok()
  File "/odoo17/custom/addons/odooplm/plm/models/ir_attachment.py", line 1312, in is_last_save_ok
    key = f"{ir_attachment_id.engineering_code}_{ir_attachment_id.engineering_revision}"
  File "/odoo17/odoo/odoo/fields.py", line 1180, in __get__
    recs._fetch_field(self)
  File "/odoo17/odoo/odoo/models.py", line 3758, in _fetch_field
    self.fetch(fnames)
  File "/odoo17/odoo/odoo/models.py", line 3793, in fetch
    query = self.with_context(active_test=False)._search([('id', 'in', self.ids)])
  File "/odoo17/custom/addons/odooplm/plm/models/ir_attachment.py", line 2304, in _search
    ids = super(models.Model, self)._search(args, offset=offset, limit=limit, order=order,
TypeError: BaseModel._search() got an unexpected keyword argument 'count'

The above server error caused the following client error:
RPC_ERROR: Odoo Server Error
    at makeErrorFromResponse (http://143.110.154.9:8069/web/assets/bacc0ba/web.assets_web.min.js:2867:163)
    at XMLHttpRequest.<anonymous> (http://143.110.154.9:8069/web/assets/bacc0ba/web.assets_web.min.js:2871:13)

FalkoG123, 2023-11-16:
Can be closed, was my error.

MaceoT, 2024-01-26:
Hi, I have the same issue, could you explain how you resolve it ? Thanks !

FalkoG123, 2024-01-26:
Hi, you need to specify clone branch for odoo 17. Default was still 16 when I was doing it.

OmniaGit, 2024-03-07:
The default branch is set to 17.0 now

closed 2024-03-07 by OmniaGit