OmniaGit/odooplm #83: Solid Works gets stuck

droidosergioh, 2024-09-20:
When I try so save document file into Odoo, Solid Works gets stuck

[image]

Steps to reproduce it:

1. Open solid works (v2021 in my case)
2. Open any .SLDASM or .SLDPRT file
3. Press into "Save to Odoo" Button
4. Solid works gets forzen. Eventually, I need to close it with the Task manager because is totally stuck

Document file should be uploaded into odoo without problem. Solid Works should work fine.

Desktop:

- OS: Windows 10
- Browser Chrome
- Version: Odoo v17, and Solid Works 2021

OmniaGit, 2024-09-20:
Dear,
did you make the login to odoo first ?

https://odooplm.omniasolutions.website/login#Login

regards,

droidosergioh, 2024-09-20:
Hello

Yes, that was the first thing I did and it was able to connect successfully.

Kindest Regards

OmniaGit, 2024-09-20:
which client version are you using ?

[image]

also can  you do e test with a file name like test.sldprt

regards.

droidosergioh, 2024-09-20:
The OdooPlm_03012024_101548

OmniaGit, 2024-09-20:
please use the latest stable version that you can find at

https://sourceforge.net/projects/openerpplm/files/Client/Stable/OdooPlm_19092024_164558.exe/download

regards,

droidosergioh, 2024-09-20:
Nice! Thank you. I got to press save, but now it gets stuck here:

[image]

Some documents were uploaded though

OmniaGit, 2024-09-20:
in order to correctly upload your model you need to fill up document data and product data.
this is made automatically by using the edit Parts command
https://odooplm.omniasolutions.website/login#Edit-Parts
once you have insert the description and the engineering code you are ready for saving your model

[image]

another way to save the model is to use the advanced saving form.
have a look at this video for more information

https://youtu.be/ZXSIR0Lvwbs

regards

OmniaGit, 2024-09-24:
I'll close the issue if you still have the problem let me know
regards

droidosergioh, 2024-09-24:
> I'll close the issue if you still have the problem let me know regards

Sure! Thank you for the help