OmniaGit/odooplm #84: Error with the Upgrade from Tree tag to List tag

sansanjo-odoo-sh-admin, 2024-12-28:
**Describe the bug** A clear and concise description of what the bug is.
When I Install the PLM module everything goes right. But when try to enable the "Enable Eco /Ecr" setting I get the next error:

```
Traceback (most recent call last):
  File "/home/odoo/src/odoo/odoo/tools/cache.py", line 103, in lookup
    r = d[key]
        ~^^^^^
  File "/usr/lib/python3/dist-packages/decorator.py", line 232, in fun
    return caller(func, *(extras + args), **kw)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/odoo/src/odoo/odoo/tools/func.py", line 97, in locked
    return func(inst, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/odoo/src/odoo/odoo/tools/lru.py", line 33, in __getitem__
    a = self.d[obj]
        ~~~~~~^^^^^
KeyError: ('ir.model.data', <function IrModelData._xmlid_lookup at 0x7f8ce9daa980>, 'mail.mail_activity_view_list')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/odoo/src/odoo/odoo/tools/convert.py", line 537, in _tag_root
    f(rec)
  File "/home/odoo/src/odoo/odoo/tools/convert.py", line 400, in _tag_record
    f_val = self.id_get(f_ref, raise_if_not_found=nodeattr2bool(rec, 'forcecreate', True))
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/odoo/src/odoo/odoo/tools/convert.py", line 519, in id_get
    res = self.model_id_get(id_str, raise_if_not_found)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/odoo/src/odoo/odoo/tools/convert.py", line 525, in model_id_get
    return self.env['ir.model.data']._xmlid_to_res_model_res_id(id_str, raise_if_not_found=raise_if_not_found)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/odoo/src/odoo/odoo/addons/base/models/ir_model.py", line 2245, in _xmlid_to_res_model_res_id
    return self._xmlid_lookup(xmlid)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/decorator.py", line 232, in fun
    return caller(func, *(extras + args), **kw)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/odoo/src/odoo/odoo/tools/cache.py", line 110, in lookup
    value = d[key] = self.method(*args, **kwargs)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/odoo/src/odoo/odoo/addons/base/models/ir_model.py", line 2238, in _xmlid_lookup
    raise ValueError('External ID not found in the system: %s' % xmlid)
ValueError: External ID not found in the system: mail.mail_activity_view_list
```

Which means that when the `mail.activity.view.list.activity.validation` view trys to inherit from the `mail.mail_activity_view_list`, Odoo doesn't find the later view.

**To Reproduce** Steps to reproduce the behavior:

1. Go to 'settings' of the PLM module
2. Click on 'Enable Eco /Ecr'
3. Click on 'Save'
4. See error

**Expected behavior**
Install the enabled settings, without error.

**Desktop (please complete the following information):**

- OS: Odoo SH
- Browser: Chrome
- Version: Odoo version 18

**Additional context**
I'm running the code on Odoo SH, on Odoo version 18 in the latest revision.

sansanjo-odoo-sh-admin, 2024-12-28:
I found that Odoo changed the `<tree>` Tag to `<list>` Tag. And so the code is correctly edited with the now new  `<list>` Tags.

But the problem is that the external Ids remain with the same name, `mail_activity_view_tree` instead of `mail_activity_view_list `. And I imagine that there was a type where every **tree** word was replaced by a **list** word. And so the tags where correctly changed but the external IDs where incorrectly changed.

So my recommendation would be to change the external IDs of the views back to the currently correct name.

What do you think?

OmniaGit, 2025-01-02:
Dear @sansanjo-odoo-sh-admin the plm  activity module is not yet ported to the v18 version, it will be done in few days.
regards

closed 2025-01-02 by OmniaGit