OmniaGit/odooplm #86: CustomProcedure File throwing an error

khushboopks, 2025-04-07:
When we use CustomProcedure file in odoo v18, solidowrks throw an error, which block the further process. error is listed below. can we get v18 customPrrocedure file.

u'(-2147352567, \'Exception occurred.\', (0, u\'Python COM Server Internal Error\', u\'Unexpected Python Error: Traceback (most recent call last):\\n  File "win32com\\\\server\\\\policy.pyc", line 277, in _Invoke_\\n  File "win32com\\\\server\\\\policy.pyc", line 282, in _invoke_\\n  File "win32com\\\\server\\\\policy.pyc", line 585, in _invokeex_\\n  File "plmintegration\\\\utils.pyc", line 259, in func_wrapper\\nTypeError: solidworksIntegrationHook__getProperties() got an unexpected keyword argument \\\'writePropsInDocumentTab\\\'\\n\', None, 0, -2147467259), None)'Traceback (most recent call last):
  File "openclerk.pyc", line 196, in func_wrapper
  File "openclerk.pyc", line 2126, in Save
  File "openclerk.pyc", line 2199, in commonSave
  File "openclerk.pyc", line 1727, in _CheckOut
  File "<COMObject OpenErpPlm.Application>", line 2, in CanCheckOut
com_error: (-2147352567, 'Exception occurred.', (0, u'Python COM Server Internal Error', u'Unexpected Python Error: Traceback (most recent call last):\n  File "win32com\\server\\policy.pyc", line 277, in _Invoke_\n  File "win32com\\server\\policy.pyc", line 282, in _invoke_\n  File "win32com\\server\\policy.pyc", line 585, in _invokeex_\n  File "plmintegration\\utils.pyc", line 259, in func_wrapper\nTypeError: solidworksIntegrationHook__getProperties() got an unexpected keyword argument \'writePropsInDocumentTab\'\n', None, 0, -2147467259), None)

khushboopks, 2025-04-07:
can i get CustomProcedure file for odoo v18?

khushboopks, 2025-04-07:
yes, previously we were working with odoo v13  and now we are migrating to v18. 

we are using plm_client_customprocedure module , as it suggest to upload customprocedure file , so we have that file in v13 but don't have any reference for v18 .   so is there any reference file for this module.

OmniaGit, 2025-04-07:
In v18 we made big change on the client application, the code that is giving you the a problem is running on the client application (Solidworks) no in the odoo server.

In order to fix the problem you need to know the change that we have made on the client application that is not open source.
Unfortunately I'm not able to give you any help on that without maintenance.

regards

closed 2025-04-07 by OmniaGit