OmniaGit/odooplm #87: Product category create throw an error in solidworks using odoo v18

khushboopks, 2025-04-07:
Error as below : 

View detailsTraceback (most recent call last):
  File "/usr/lib/python3/dist-packages/odoo/addons/base/controllers/rpc.py", line 151, in xmlrpc_1
    response = self._xmlrpc(service)
               ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/odoo/addons/base/controllers/rpc.py", line 139, in _xmlrpc
    result = dispatch_rpc(service, method, params)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/odoo/http.py", line 398, in dispatch_rpc
    return dispatch(method, params)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/odoo/service/model.py", line 39, in dispatch
    res = execute_kw(db, uid, *params[3:])
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/odoo/service/model.py", line 61, in execute_kw
    return execute(db, uid, obj, method, *args, **kw or {})
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/odoo/service/model.py", line 68, in execute
    res = execute_cr(cr, uid, obj, method, *args, **kw)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/odoo/service/model.py", line 52, in execute_cr
    result = retrying(partial(odoo.api.call_kw, recs, method, args, kw), env)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/odoo/service/model.py", line 137, in retrying
    result = func()
             ^^^^^^
  File "/usr/lib/python3/dist-packages/odoo/api.py", line 500, in call_kw
    raise AttributeError(f"The method '{name}' does not exist on the model '{model._name}'")
AttributeError: The method 'name_get' does not exist on the model 'product.category'

OmniaGit, 2025-04-07:
can you give us more information in how you get this error ?
which client version are you using ?

regards

khushboopks, 2025-04-07:
When we are trying to create new product category from the solidworks at that time we are facing this error.
we are using 2024 solidworks.

OmniaGit, 2025-04-07:
I mean which odoo  plm client version are  you using ?
regards,

khushboopks, 2025-04-11:
Client version : 18032025_232838

OmniaGit, 2025-04-23:
Try to create the name_get function on the product_category object.
Just to check if this is the problem.

in the meantime we do some test internally to see if we reproduce the problem

regards

khushboopks, 2025-05-07:
Okay , Thank you . 
is that possible to test in any other client version? if you can suggest any client version so i can test in the same.